nrf-rs/microbit #1: Howto Guide: How to use this crate!

mattgathu, 2018-05-14:
I'm trying to use this crate and have rust on the micro:bit but I don't know how to proceed beyond compiling the code.

Could you update the README to have instructions on how to use the crate? 🙂 

I'm also happy to update it if you can give me pointers or notes on how to get started with this great crate. Thanks.

therealprof, 2018-05-14:
You can compile the examples with `cargo build --release --examples`. This will generate the binaries in `target/thumbv6m-none-eabi/release/examples/`. I'm not sure whether you can flash those using drag'n'drop (at least on my Mac that's a bit hit and miss) but if you have `openocd` installed you should be able to use the included `openocd_program.sh` script to flash the binaries to the Micro:Bit, like
```
# ./openocd_program.sh target/thumbv6m-none-eabi/release/examples/gpio_hal_ledbutton
Open On-Chip Debugger 0.10.0+dev-00360-gffd6b78a (2018-03-17-23:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 1.10
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bb11477
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Listening on port 3333 for gdb connections
Info : nRF51822-QFAA(build code: H0) 256kB Flash
flash 'nrf51' found at 0x00000000
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0x000000a8 msp: 0x20004000
** Programming Started **
auto erase enabled
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg/nrf52.cfg to disable it
wrote 1024 bytes from file target/thumbv6m-none-eabi/release/examples/gpio_hal_ledbutton in 0.205990s (4.855 KiB/s)
** Programming Finished **
** Verify Started **
verified 362 bytes in 0.085142s (4.152 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

cathalgarvey, 2018-05-16:
Hey @therealprof - Thanks for the above!

But, I'm having trouble even getting the environment set up for Cortex M, it seems the instructions I find by searching are outdated, and the toolchain is not available via rustup. Is there any way you could point me/us at a more up to date guide, to go from x86 rust nightly on Ubuntu to having a working toolchain so that `cargo build --release --examples` works? :)

Thank you!

therealprof, 2018-05-16:
@cathalgarvey Have you checked https://github.com/rust-lang-nursery/embedded-wg/blob/master/newsletters/2018-05-13-the-path-towards-stable.md ?

Note that you might need a different toolchain, Cortex-M0 (used e.g. by the Micro:bit) would be `thumbv6m-none-eabi`.

cathalgarvey, 2018-05-16:
:man_facepalming: It turns out I just needed to `rustup update`; the required toolchain wasn't known by that triplet to my install, and I was getting cryptic errors.

The `cargo build --release --examples` now finishes successfully. I can also confirm that when I drag/drop an example onto the Microbit, it appears to flash and then work as expected. Thanks a million, this is a huge upgrade to my Microbits! :)

therealprof, 2018-05-31:
@mattgathu @cathalgarvey I've just started a new blog post series about Programming in Rust for the Micro:bit here: https://www.eggers-club.de/blog/2018/05/31/rust-on-the-microbit-101-part-1/

At the moment it only covers setting up the toolchain but I'll put up more posts as time permits. Please feel free to send me your wishes what to cover next.

closed 2018-05-31 by therealprof