nrf-rs/microbit #17: ERROR(cortex-m-rt): The interrupt vectors are missing.

oskarhagberg, 2019-07-17:
Trying out your tutorial but gettings stuck compiling when adding `microbit` to the .toml

`ERROR(cortex-m-rt): The interrupt vectors are missing.`

oskarhagberg, 2019-07-17:
Finally got it to build after adding `extern crate microbit;` to `main.rs`

What gives? 😄

robyoung, 2021-05-11:
Is this still an issue?

jgarth, 2021-05-31:
I encountered the same issue just now using the latest version of both `microbit` and `cortex-m-rt`. Adding the `extern crate` statement mentioned above worked for me as well.

jwhittle933, 2022-07-20:
This also happens if you don't specify the  chip in Embed.toml.
```
[default.general]
# chip = "nrf52833_xxAA" # uncomment this line for micro:bit V2
# chip = "nrf51822_xxAA" # uncomment this line for micro:bit V1
```