nrf-rs/microbit #18: Soundness-hole in NVIC::enable

peacememories, 2019-11-11:
When building the crate the compiler emits deprecation warnings for `cortex_m::peripheral::nvic::<impl cortex_m::peripheral::NVIC>::enable`, because it should apparently be unsafe. The replacement is `NVIC::unmask`, which is an associated function instead of a method, and is masked unsafe. It doesn't seem to be a straight find-replace fix though.

therealprof, 2019-11-11:
Huh? It's pretty straight forward, e.g. (including additional cleanup after the change):
```
--- a/examples/gpio_hal_printbuttons.rs
+++ b/examples/gpio_hal_printbuttons.rs
@@ -9,7 +9,6 @@ use microbit::hal::serial;
 use microbit::hal::serial::BAUD115200;

 use cortex_m::interrupt::Mutex;
-use cortex_m::peripheral::Peripherals;
 use cortex_m_rt::entry;

 use core::cell::RefCell;
@@ -21,10 +20,10 @@ static TX: Mutex<RefCell<Option<serial::Tx<UART0>>>> = Mutex::new(RefCell::new(N

 #[entry]
 fn main() -> ! {
-    if let (Some(p), Some(mut cp)) = (microbit::Peripherals::take(), Peripherals::take()) {
+    if let Some(p) = microbit::Peripherals::take() {
         cortex_m::interrupt::free(move |cs| {
             /* Enable external GPIO interrupts */
-            cp.NVIC.enable(microbit::Interrupt::GPIOTE);
+            unsafe { microbit::NVIC::unmask(microbit::Interrupt::GPIOTE) };
             microbit::NVIC::unpend(microbit::Interrupt::GPIOTE);

             /* Split GPIO pins */
```

therealprof, 2019-11-11:
Of course we could also take the NVIC directly from `cortex-m` instead of re-exporting it in this crate.

peacememories, 2019-11-11:
Yeah, replacing the calls is easy enough, but at the same time this creates about 6 new unsafe blocks in the codebase, which need to be checked for soundness (as in, am I allowed to write to the interrupt mask right now?)
I think this would greatly benefit from a safe abstraction.

therealprof, 2019-11-11:
It is only unsafe when used with mask-based critical sections so it's a very theoretical safety issue and one which certainly doesn't apply to these examples.

The only "safe" option to abstract over the unsafe interrupt setup would be to have a crate which handles to complete early entry MCU setup, especially the interrupt setup. One crate which does is is `cortex-m-rtfm`.

peacememories, 2019-11-11:
Yeah you're right.
I already have a commit with all the changes in question on my laptop from earlier, and could create a PR after I'm home from work if you'd like me to.

therealprof, 2019-11-11:
Sure, PRs are always welcome.

PR #19 (2020-01-07): Replaced NVIC::enable with NVIC::unmask
This removes all usages of the deprecated `NVIC::enable` and replaces them with `NVIC::unmask`.
According to upstream `NVIC::enable` has a soundness issue and should be marked unsafe, which is why it was deprecated.

Fixes #18
closed 2020-01-07 by therealprof