nrf-rs/microbit #21: Taking one peripheral at a time?

Ben-Lichtman, 2019-12-18:
Is there any way to take one peripheral at a time? When I take them all I get issues with having to move UART0 out for serial_port which prevents me from borrowing the rest of the struct.

therealprof, 2019-12-18:
Not at the moment, cf. https://github.com/therealprof/microbit/blob/37550389870373505d7b844f80f8d7ad2b3f108f/src/lib.rs#L15

The abstraction at the moment just caters for the UART connected to the debugger. You should be able to create multiple UARTs using the HAL directly, though.

mattheww, 2020-01-07:
When I ran into this problem I ended up defining several pin-owning structs and a 'split' method to distribute ownership.

https://github.com/mattheww/rmicrobit/blob/master/src/gpio.rs
https://github.com/mattheww/rmicrobit/blob/master/src/display/display_port.rs

@therealprof, would you be interested in some form of this code for the microbit crate?

therealprof, 2020-01-07:
looks great!

mattheww, 2021-05-15:
Yes, it's now possible to use the rest of the pins after giving some of them away via `serial_port!` and/or `display!`.

closed 2021-05-15 by robyoung