nrf-rs/microbit #23: Compile Error: microbit/src/display/doc_example.rs

Code-Tap, 2020-02-27:
[image]

Im am struggling to get this to build, as you can see by the errors it seems to have multiple issues.

```bash
rustup update
info: syncing channel updates for 'stable-x86_64-unknown-linux-gnu'
info: syncing channel updates for 'nightly-x86_64-unknown-linux-gnu'

   stable-x86_64-unknown-linux-gnu unchanged - rustc 1.41.1 (f3e1a954d 2020-02-24)
  nightly-x86_64-unknown-linux-gnu unchanged - rustc 1.43.0-nightly (abc3073c9 2020-02-26)```

my cargo.toml
```
[dependencies]
panic-halt = "~0.2"
microbit="~0.8.0"
cortex-m-rt="~0.6"
cortex-m-rtfm = "0.5.1"
```
my memory.x
```
MEMORY
{
  /* NOTE K = KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x00000000, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 16K
}
```

thanks for any help you can give to get this running

therealprof, 2020-02-28:
Hm, what is `microbit-blinky`? Judging from the error message I have a suspicion that you're using parts of the example code was written for an older version of `cortex-m-rtfm` than the version you're using. Try downgrading the version.

Code-Tap, 2020-02-28:
microbit-blinky was a tutorial i was following here :
[https://flames-of-code.netlify.com/blog/rust-microbit/](url)

but i couldent get that one to work, so i came to the crate and tried to find a working example.
Ill delete and reclone my local copy of this repo and try again.

therealprof, 2020-02-28:
There's an example for making the display work right in this crate: https://github.com/therealprof/microbit/blob/master/examples/led_nonblocking.rs

If you check out the the repository you should be able to compile it via:
```
cargo build --release --example led_nonblocking
```
and then we can take it from there?

therealprof, 2020-02-28:
I've now moved the example from the doc_example directly to an example and added the correct dependency to the `dev-dependencies` so this can be built out of the box.

https://github.com/therealprof/microbit/blob/master/examples/led_rtfm.rs

Hope that helps.

Code-Tap, 2020-02-28:
Thats brilliant!
Thanks for responding so quickly. I will have a go later when im back to my computer.
I cant wait to do some speed comparisons compared to microython.

therealprof, 2020-02-28:
No problem.

Code-Tap, 2020-02-28:
Huzzah!!!

```bash
~/Projects/rustlang
❯ cd microbit

microbit on  master is 📦 v0.8.0 via 🦀 v1.41.1
❯ cargo build
    Updating crates.io index
   Compiling semver-parser v0.7.0
   Compiling typenum v1.11.2
   Compiling proc-macro2 v1.0.9
   Compiling unicode-xid v0.2.0
   Compiling cortex-m v0.6.2
   Compiling syn v1.0.16
   Compiling stable_deref_trait v1.1.1
   Compiling cortex-m-rt v0.6.12
   Compiling vcell v0.1.2
   Compiling fpa v0.1.0
   Compiling r0 v0.2.2
   Compiling nrf51 v0.7.0
   Compiling void v1.0.2
   Compiling nb v0.1.2
   Compiling microbit v0.8.0 (~/Projects/rustlang/microbit)
   Compiling tiny-led-matrix v1.0.2
   Compiling semver v0.9.0
   Compiling volatile-register v0.2.0
   Compiling embedded-hal v0.2.3
   Compiling rustc_version v0.2.3
   Compiling quote v1.0.2
   Compiling generic-array v0.13.2
   Compiling generic-array v0.12.3
   Compiling bare-metal v0.2.5
   Compiling cast v0.2.3
   Compiling as-slice v0.1.3
   Compiling aligned v0.3.2
   Compiling cortex-m-rt-macros v0.1.8
   Compiling nrf51-hal v0.7.1
    Finished dev [unoptimized + debuginfo] target(s) in 52.06s

microbit on  master is 📦 v0.8.0 via 🦀 v1.41.1 took 52s
❯
```

therealprof, 2020-02-28:
Well, you haven't built any examples yet. ;)
Try:
```
cargo build --release --examples
```

Code-Tap, 2020-02-28:
Another success!
found the built files here
```
microbit/target/thumbv6m-none-eabi/release/examples
```

Just got to read up on transferring the code to the microbit and running it now!
did'nt expect to get this far this fast!

Code-Tap, 2020-02-28:
I feel like I have superpowers tonight
You've been amazing and I feel I'm now in a place where I can really start learning about rust and embedded.

[image]

closed 2020-02-28 by Code-Tap