nrf-rs/microbit #35: Move to the nrf-rs org

robyoung, 2021-04-21:
This was mentioned in #27 

Is there anything that should be done before this happens?

therealprof, 2021-04-21:
I think what should be done:

- [ ] Update the documentation and general files
- [ ] Add GitHub Actions CI
- [ ] Release the current state

Are you already lurking on the matrix channel by any chance so we can coordinate? It's at in case you're not: https://app.element.io/#/room/#nrf-rs:matrix.org

robyoung, 2021-04-26:
Things I've addressed:
- Update the documentation and general files in #36 
- Add GitHub Actions CI in #28 #32 and #37

Things remaining:
- Add bors
- Release the current state

therealprof, 2021-04-26:
I've already amended the CHANGELOG.md with changes link for last release vs unreleased.

If you have time to do a PR bumping the version number, adding the release version line (and the link to the diffs 😉) and maybe also add a note that CHANGELOG keeping started with version 0.8.0, then I think we can do the release, clean up the issue list and move it over. 

We can/must do bors and variety of other changes on the repo but also it's settings anyway so that can be done after the move.

therealprof, 2021-04-29:
@robyoung Would you prefer me to prepare the release?

robyoung, 2021-04-29:
Sorry, I completely missed this.

therealprof, 2021-04-29:
And we have moved... 🎉

closed 2021-04-29 by therealprof