nrf-rs/microbit #73: Examples Display Nonblocking & RTIC broken

caemor, 2021-11-10:
The update in https://github.com/nrf-rs/microbit/commit/7c322400213a15c087fbe433306e7c9aaece1b04 broke the non blocking display examples.

I haven't found the time yet to debug which part of the change is causing the issue. But nothing is displayed with the newer version in contrast to the older version where everything works, where the main difference seems to be the clock function.

caemor, 2021-11-10:
Thanks. This fixes the issue for the nonblocking example for me as well. Do you have an idea how this might apply to the rtic example as well?

cathiele, 2021-11-11:
> Thanks. This fixes the issue for the nonblocking example for me as well. Do you have an idea how this might apply to the rtic example as well?

In https://github.com/cathiele/microbit/blob/display-rtic-fix/examples/display-rtic/src/main.rs i did a quick hack for fixing it for the V2 board. But this is not a nice solution.

caemor, 2021-11-11:
Another solution is to modify the Board::take function to steal the Core Peripherals which also isn't the best way to go forward. I am going to ask the RTIC people if there is better solution to this problem.

cathiele, 2021-11-11:
I think making ```Board::new(p, cp)``` public is the best solution for this.

commit 87a0b5d2 (2021-11-11): Merge #76

76: Make `Board::new(p, cp)` pub and fix RTIC example r=robyoung a=caemor

Fixes #73 thanks to the help of `@korken89` in the rtic matrix.

RTIC takes the core peripherals already for its own context. There board should take these directly from the context for its own creation.
PR #76 (2021-11-11): Make `Board::new(p, cp)` pub and fix RTIC example
Fixes #73 thanks to the help of @korken89 in the rtic matrix.

RTIC takes the core peripherals already for its own context. There board should take these directly from the context for its own creation.
closed 2021-11-11 by bors[bot]