nrf-rs/microbit #78: Text display support

mattheww, 2021-11-13:
The C++ micro::bit runtime includes a 5x5 font, and support for scrolling text display.

I wrote some Rust code to do the same in 2019 (it's the main reason why the display code goes via a `Render` trait rather than just expecting a bitmap).

I copied the 'pendolino3' font from [`microbit-dal`][1], which is MIT-licensed.

You can see the original implementaion in [`rmicrobit`][2] (though it can be improved now const generics are available).

I think it would be nice for the microbit crate to have at least an example showing how to do text, and I've now got some time to look at these things again.

Do you think it would make more sense to add this support to the `display::nonblocking` module (parallel to `image.rs`), or to do this in a separate crate like `tiny-led-matrix`?

At present the font code returns `BitImage`s, which are defined in `microbit`'s `image.rs`.

[1]: https://github.com/lancaster-university/microbit-dal/blob/master/source/core/MicroBitFont.cpp

[2]: https://github.com/mattheww/rmicrobit/tree/master/src/graphics/

robyoung, 2021-11-15:
Text display support would be amazing!

I personally think it would be best in a separate crate with an example in this crate. I really like the batteries included approach you get with the official micro:bit runtimes. They certainly make getting started a little easier. I feel like one of the goals of this project should be to introduce people who may not be familiar with embedded programming or Rust programming to the embedded Rust ecosystem and it's norms. One of those norms seems to be small libraries that solve specific problems and can be plugged together.

If we do go with separate crates I think it would be good to figure out a good way of directing users towards other crates that work well on micro:bit and they may be interested in.

What are your thoughts?

mattheww, 2021-11-15:
That seems reasonable.

Making a separate crate is easy enough; the one complication is getting at `BitImage`.

If I understand correctly, `microbit-common` won't compile without specifying either the `v1` or the `v2` feature.

So if I made a new crate depend on `microbit-common` so it could get at `BitImage`, I think the new crate would itself need `v1` and `v2` features, even though nothing in it should care about the micro:bit version.

I think to start with the expedient thing will be to duplicate `BitImage` in the new crate; then we could think about the best place for it to live.

robyoung, 2021-11-15:
With const generics could `BitImage` and `GreyscaleImage` live in tiny-led-matrix?

mattheww, 2021-11-16:
I don't think Rust is quite up to doing `BitImage::new` for an arbitrary size yet. I'd want `array::from_fn` but that's not `const`, and currently nightly-only anyway.

mattheww, 2021-11-24:
I don't have any cleverer ideas for what to do with `BitImage`, and it's not a large piece of duplication (most of the code disappears at runtime), so let's just close this.

closed 2021-11-24 by mattheww