nrf-rs/microbit #82: Example fails to run

wpd, 2021-12-26:
Following the example given at [crates.io](https://crates.io/crates/microbit) on MaoOS, I see the following:
```
$ cargo run --release --manifest-path ./examples/display-blocking/Cargo.toml --features v2 --target thumbv7em-none-eabihf
    Finished release [optimized] target(s) in 0.25s
     Running `probe-run --chip nRF52833_xxAA target/thumbv7em-none-eabihf/release/display-blocking`
Error: defmt wire format version mismatch: firmware is using 0.2, `probe-run` supports 3
suggestion: `cargo install` a different version of `probe-run` that supports defmt 0.2
````
Can you please help me learn how to `cargo install` a different version of `probe run` that supports defmt 0.2.
--wpd

wpd, 2021-12-26:
One way to work around this is:
```
$ cargo install --version "0.2.0" probe-run
```
I don't know if you want to update the documentation at [crates.io](https://crates.io/crates/microbitl), or update all of the `Cargo.toml` files in your repository, and update the `rubble` repository to use defmt 0.3.0.

drewhk, 2022-01-27:
I have the same issue, but trying the workaround above got into another error (while installing probe-run 0.2)
```
  --- stdout
  Can't find libusb pkg: VcpkgNotFound("No vcpkg installation found. Set the VCPKG_ROOT environment variable or run 'vcpkg integrate install'")
```

shaggygi, 2022-04-15:
I'm having same issues.  Any update on how to fix?  Thx

therealprof, 2022-05-17:
The problem is that the `ble-beacon` example uses `rubble` which has a hardcoded dependency on `defmt` in version `0.2.0`.

In order to upgrade `defmt` the two options are: Someone updates `rubble` or we remove the `ble-beacon` example.

caemor, 2022-07-21:
Should be resolved by #88

therealprof, 2022-07-21:
Indeed, sorry for not noticing that this issue can be closed.

closed 2022-07-21 by therealprof