nrf-rs/microbit #108: Non-blocking display API affects wrong LEDs

br0kenpixel, 2023-03-04:
Hi. I've been experimenting with the non-blocking display API and noticed that in some cases the wrong LEDs are being affected.
For example, if I take the examples/display-nonblocking example and change the `heart_image` function to display a smiley face instead:
```rust
fn heart_image(inner_brightness: u8) -> GreyscaleImage {
    let b = inner_brightness;
    GreyscaleImage::new(&[
        [0, 0, 0, 0, 0],
        [0, b, 0, b, 0],
        [0, 0, 0, 0, 0],
        [b, 0, 0, 0, b],
        [0, b, b, b, 0],
    ])
}
```
The 4th and 14th LED (the ones above and below the right eye) are also turning on, although very faintly. If I try to display the same image using the blocking API, it looks fine.

> Micro:Bit version: 2.1

mattheww, 2024-01-31:
I can reproduce this behaviour with a debug build, but it's OK in a release build.

Perhaps the debug build isn't running fast enough to get all the work done before the next interrupt comes in.

mattheww, 2024-02-01:
I've had a look and I don't see why a dropped interrupt would lead to this symptom.

mattheww, 2024-02-03:
I think I see the cause: on the micro:bit v2 the GPIO pins are on two ports, and we weren't being careful enough about the order of writes to make sure that intermediate states don't cause unwanted LEDs to be lit.

mattheww, 2024-02-03:
I had a quick look at the blocking code. That writes pins individually rather than using `OUTSET` and `OUTCLR`.

The order it uses looks safe to me: it explicitly sets all columns 'dark' after it's finished displaying a row, then sets the new row 'light', then starts setting columns 'light'.

PR #129 (2024-02-03): Non-blocking display: change the order of GPIO writes
Closes #108

On the micro:bit v2 the GPIO pins are on two ports, and it takes four IO writes to update all the pins: one "clear multiple pins" and one "set multiple pins" per port.

(It's possible to set pins to both low and high using `OUT`, but we can't use that as we don't own all the pins on the port.)

So there is a brief period where the pins are in a mix of the old and new states, and evidently in debug builds this can persist for long enough to be visible.

We can fix this by changing the order of writes: make the first write be the [... cut]
closed 2024-02-03 by lulf