nrf-rs/microbit #110: failed to resolve: use of undeclared crate or module `microbit`

curt-wise, 2023-03-24:
I have a micro:bit v2.21 and using Linux Kubuntu, i created a new rust project using cargo and modified the files like this:

`Cargo.toml`

```rust
[package]
name = "test_0"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
microbit-v2 = { version = "0.13.0", optional = true }
cortex-m = { version = "0.7.7", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.3"
panic-halt = "0.2.0"
defmt-rtt = "0.4.0"
defmt = "0.3.2"

[features]
v2 = ["microbit-v2"]

default = ["defmt-default"]

# do NOT modify these features
defmt-default = []
defmt-trace = []
defmt-debug = []
defmt-info = []
defmt-warn = []
defmt-error = []
```

`main.rs`

```rust
#![no_std]
#![no_main]

use defmt_rtt as _;
use panic_halt as _;

use cortex_m_rt::entry;

use microbit::{
    board::Board,
    display::blocking::Display,
    hal::{prelude::*, Timer},
};

#[entry]
fn main() -> ! {
    if let Some(board) = Board::take() {
        let mut timer = Timer::new(board.TIMER0);
        let mut display = Display::new(board.display_pins);

        #[allow(non_snake_case)]
        let letter_I = [
            [0, 1, 1, 1, 0],
            [0, 0, 1, 0, 0],
            [0, 0, 1, 0, 0],
            [0, 0, 1, 0, 0],
            [0, 1, 1, 1, 0],
        ];

        let heart = [
            [0, 1, 0, 1, 0],
            [1, 0, 1, 0, 1],
            [1, 0, 0, 0, 1],
            [0, 1, 0, 1, 0],
            [0, 0, 1, 0, 0],
        ];

        #[allow(non_snake_case)]
        let letter_R = [
            [0, 1, 1, 0, 0],
            [0, 1, 0, 1, 0],
            [0, 1, 1, 0, 0],
            [0, 1, 0, 1, 0],
            [0, 1, 0, 1, 0],
        ];

        #[allow(non_snake_case)]
        let letter_u = [
            [0, 0, 0, 0, 0],
            [0, 0, 0, 0, 0],
            [0, 1, 0, 1, 0],
            [0, 1, 0, 1, 0],
            [0, 1, 1, 1, 0],
        ];

        #[allow(non_snake_case)]
        let letter_s = [
            [0, 0, 0, 0, 0],
            [0, 0, 1, 1, 0],
            [0, 1, 0, 0, 0],
            [0, 0, 1, 0, 0],
            [0, 1, 1, 1, 0],
        ];

        #[allow(non_snake_case)]
        let letter_t = [
            [0, 0, 1, 0, 0],
            [0, 1, 1, 1, 0],
            [0, 0, 1, 0, 0],
            [0, 0, 1, 0, 0],
            [0, 0, 1, 0, 0],
        ];
        loop {
            display.show(&mut timer, letter_I, 1000);
            display.show(&mut timer, heart, 1000);
            display.show(&mut timer, letter_R, 1000);
            display.show(&mut timer, letter_u, 1000);
            display.show(&mut timer, letter_s, 1000);
            display.show(&mut timer, letter_t, 1000);
            display.clear();
            timer.delay_ms(250_u32);
        }
    }

    panic!("End");
}
```

When i run `cargo embed --target thumbv7em-none-eabihf`, i get that error:
```bash
error: could not compile `test_0` due to 3 previous errors
error[E0433]: failed to resolve: use of undeclared crate or module `microbit`
 --> src/main.rs:9:5
  |
9 | use microbit::{
  |     ^^^^^^^^ use of undeclared crate or module `microbit`

error[E0432]: unresolved imports `microbit::board::Board`, `microbit::display::blocking::Display`, `microbit::hal::Timer`
  --> src/main.rs:10:5
   |
10 |     board::Board,
   |     ^^^^^^^^^^^^
11 |     display::blocking::Display,
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^
12 |     hal::{prelude::*, Timer},
   |                       ^^^^^

error: aborting due to 2 previous errors

Some errors have detailed explanations: E0432, E0433.
For more information about an error, try `rustc --explain E0432`.
       Error Failed to run cargo build: exit code = Some(101).
```

Nequo, 2023-05-25:
Have you tried running `cargo embed --target thumbv7em-none-eabihf --features v2`?
Looks like the [docs](https://github.com/nrf-rs/microbit#run-an-example) use `--features v2` in the example cargo run command.

Nequo, 2023-05-29:
@curt-wise does the thumbs-up mean that it worked for you? If so I think we should close this issue 🙂

curt-wise, 2023-05-29:
> Have you tried running `cargo embed --target thumbv7em-none-eabihf --features v2`? Looks like the [docs](https://github.com/nrf-rs/microbit#run-an-example) use `--features v2` in the example cargo run command.

Yes i know that the docs state that. But i want to make it work using only:
`cargo embed --target thumbv7em-none-eabihf`
and preferably using only:
`cargo embed`

Any idea how i might do that ?

Nequo, 2023-05-29:
@curt-wise Features you can read up on [cargo features](https://doc.rust-lang.org/cargo/reference/features.html) but as I understand it you would have to create your own version of the crate with [all conditional compilation for "v2"](https://github.com/search?q=repo%3Anrf-rs%2Fmicrobit++%23%5Bcfg%28feature+%3D+%22v2%22%29%5D&type=code) removed. 

See [here](https://doc.rust-lang.org/cargo/reference/config.html#buildtarget) on how to set a default target.

curt-wise, 2023-05-29:
I wanted to create my own project from scratch but could not do that. Like doing `cargo new <project_name>` and add the microbit v2 crate in `cargo.toml`. I wanted to do that instead of cloning the repository and modifying in it.

nocommentsgood, 2023-08-05:
possibly have to remove the optional:
[dependencies]
microbit-v2 = { version = "0.13.0", optional = true }

remove from Cargo.toml:
[features]
v2 = ["microbit-v2"]

try to compile. if compilation doesn't work, check errors, may have to add this to your .cargo/config if you get linker errors:

[target.'cfg(all(target_arch = "arm", target_os = "none"))']
rustflags = [ "-C", "link-arg=-Tlink.x",
]

should be able to create a new project with cargo new and flash with cargo embed --target thumbv7em-none-eabihf