nrf-rs/microbit #111: The port pins on the edge connector are different for v1 and v2

tsadowski, 2023-04-15:
Hi,
I have been working on a Servo example. After it worked on a v1 board I tried it on a v2 board a well. I could compile and flash and it did run but the servos were not working. After some fiddling around I found out I needed to change the pins.

I don't know, what the best solution is. My preferred way would be to change the names of the unused pins in board.rs to the edge connector pin names. But this would be a breaking change.

What do you think?

Best Regards,
Torsten

Micro:Bit Edge Pin | Port Pins v1 | Port Pins v2
-- | -- | --
Pad 0 | P0.03 | P0.02
Pad 1 | P0.02 | P0.03
Pad 2 | P0.01 | P0.04
3 | P0.04 | P0.31
4 | P0.05 | P0.28
5 | P0.17 | P0.14
6 | P0.12 | P1.05
7 | P0.11 | P0.11
8 | P0.18 | P0.10
9 | P0.10 | P0.09
10 | P0.06 | P0.30
11 | P0.26 | P0.23
12 | P0.20 | P0.12
13 | P0.23 | P0.17
14 | P0.22 | P0.01
15 | P0.21 | P0.13
16 | P0.16 | P1.02
19 | P0.00 | P0.26
20 | P0.30 | P1.00

BartMassey, 2024-07-30:
I like the concept of this change, but it's a pretty major API break.

In the meantime you can use `v1` and `v2` feature flags to kludge around it, I guess?

tsadowski, 2024-07-31:
Ups, I should have closed this long ago. I added PAD and EDGE aliases to the relevant pins and this has already been merged (PR #112). I was at this time new to Rust and coming from MakeCode, Python and Forth I had a difficult time to understand why a change from v1 to v2 did not simply work.
Right now I have to use feature flags because Adc::read_channel is different for v1 and v2.
Cheers, Torsten

tsadowski, 2024-07-31:
Fixed in PR #112

closed 2024-07-31 by tsadowski