nrf-rs/microbit #131: microbit/microbit-common/src/display/nonblocking ⸺ defect in timings?

bravequickcleverfibreyarn, 2024-02-19:
I went through example regard using nonblocking display — https://github.com/nrf-rs/microbit/blob/main/examples/display-nonblocking/src/main.rs.

I noticed that some LEDs shine very dimmly in their core that should not. This is permanent for pattern being displayed. Since I verified using blocking API there is no issue with LEDs on this, I suppose some wrong. 

My unqualified guess is there is some interferrence regard timing.

From doc.
> This uses a 6ms period to light each of the three internal LED rows, so that the entire display is updated every 18ms.

I tried some interrupt priorities and RTC0 frequencies variations but issue is resistable to this.

#### Sample of displaying letter E; RTC0 with 100 prescaler, NVIC priority 32, TIMER2 NVIC priority 64
[image]

So I wonder if there is bug somewhere seeking for remendy.

bravequickcleverfibreyarn, 2024-02-19:
Yes, it's same.

BartMassey, 2024-02-20:
@lulf Should I make a release based on this merged PR plus other stuff?