nrf-rs/microbit #133: Publish updated version

BartMassey, 2024-03-06:
At this point, it seems like a good idea to take some version of #132, bump up to just-released `nrf-hal` 0.16.1 (and any other version bumps that seem good), and tag and publish version 0.13.1 of this crate. Any objections or blockers? Otherwise I'll put in a PR…

BartMassey, 2024-03-06:
(We should check for breaking changes and maybe publish 0.17.0 instead depending on what we find.)

mattheww, 2024-03-06:
Would this release be without #123?

BartMassey, 2024-03-06:
@mattheww Yes. Version 0.16.1 (just released, thank you @qwandor !) is intended to be the last version that everybody can 100% upgrade their existing code to without changing their code or their `Cargo.toml`.

The next version will jump everything to `embedded-hal` 1.0`and stuff, which *shouldn't* break existing code itself (thank you @qwandor !), but would maybe require using versions of other things that depend on `embedded-hal` at 1.0 to get everything to build.

BartMassey, 2024-03-06:
@mattheww Yes. This release would depend on just-released `nrf-hal` 0.16.1, which shouldn't break any existing code. We would then want to upgrade everything to `nrf-hal` 0.17 once it's out and do another 0.x release of this crate.

hdoordt, 2024-06-17:
@therealprof How do you feel about a release of the current state of things?

hdoordt, 2024-06-18:
Ah yes, in that case I'd like to repurpose this issue to express my wish of yet another release, based on nrf-hal 0.18.0

PR #149 (2024-06-19): Prepare for 0.15.0 release.
Fixes #133.
closed 2024-06-19 by qwandor