nrf-rs/microbit #145: Modify imports for v2-speaker example

a-camarillo, 2024-04-24:
When building the `v2-speaker` example I am receiving both a warning and error related to `OutputPin`.

On line [43](https://github.com/nrf-rs/microbit/blob/0b579e4d4b334f888f1befc09f3bfdfde3bf4aff/examples/v2-speaker/src/main.rs#L43) I am receiving the error: `no method named 'set_low' found for struct 'P0_00' in the current scope`, as well as, an unused import for
 ```rust
 use embedded_hal::digital::OutputPin;
 ```

I was able to fix this error by removing the `embedded_hal` import and replacing it with:
```rust
use microbit::hal::prelude::OutputPin;
```

if this is indeed an error in the example code I'm happy to open up a PR for the change.

BartMassey, 2024-04-24:
Would it be possible to paste a before-and-after of `Cargo.lock`? Would be nice to know what versions are involved here.

a-camarillo, 2024-04-24:
Here is the Cargo.lock with `use embedded_hal::digital::OutputPin;` and the build failing.
<details>
<summary> Before Cargo.lock </summary>

```
# This file is automatically @generated by Cargo.
# It is not intended for manual editing.
version = 3

[[package]]
name = "az"
version = "1.2.1"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "7b7e4c2464d97fe331d41de9d5db0def0a96f4d823b8b32a2efd503578988973"

[[package]]
name = "bare-metal"
version = "0.2.5"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "5deb64efa5bd81e31fcd1938615a6d98c82eafcbcd787162b6f63b91d6bac5b3"
dependencies = [
 "rustc_version",
]

[[package]]
name = "bare-metal"
version = "1.0.0"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "f8fe8f5a8a398345e52358e18ff07cc17a568fbca5c6f73873d3a62056309603"

[[package]]
name = "bitfield"
version = "0.13.2"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "46afbd2983a5d5a7bd740ccb198caf5b82f45c40c09c0eed36052d91cb92e719"

[[package]]
name = "bitflags"
version = "1.3.2"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "bef38d45163c2f1dde094a7dfd33ccf595c92905c8f8f4fdc18d06fb1037718a"

[[package]]
name = "bytemuck"
version = "1.15.0"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "5d6d68c57235a3a081186990eca2867354726650f42f7516ca50c28d6281fd15"

[[package]]
name = "cast"
version = "0.3.0"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "37b2a672a2cb129a2e41c10b1224bb368f9f37a2b16b612598138befd7b37eb5"

[[package]]
name = "cfg-if"
version = "1.0.0"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "baf1de4339761588bc0619e3cbc0120ee582ebb74b53b4efbf79117bd2da40fd"

[[package]]
name = "cortex-m"
version = "0.7.7"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "8ec610d8f49840a5b376c69663b6369e71f4b34484b9b2eb29fb918d92516cb9"
dependencies = [
 "bare-metal 0.2.5",
 "bitfield",
 "critical-section 1.1.2",
 "embedded-hal 0.2.7",
[... 887 lines cut ...]
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "77439c1b53d2303b20d9459b1ade71a83c716e3f9c34f3228c00e6f185d6c002"

[[package]]
name = "version_check"
version = "0.9.4"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "49874b5167b65d7193b8aba1567f5c7d93d001cafc34600cee003eda787e483f"

[[package]]
name = "void"
version = "1.0.2"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "6a02e4885ed3bc0f2de90ea6dd45ebcbb66dacffe03547fadbb0eeae2770887d"

[[package]]
name = "volatile-register"
version = "0.2.2"
source = "registry+https://github.com/rust-lang/crates.io-index"
checksum = "de437e2a6208b014ab52972a27e59b33fa2920d3e00fe05026167a1c509d19cc"
dependencies = [
 "vcell",
]
```
</details>

BartMassey, 2024-04-25:
Yeah, you'll need to delete `Cargo.lock` before each build, as it's cumulative.

You can see that `Cargo.lock` currently contains both `embedded_hal` 0.2.7 and `embedded_hal` 1.0.0. This is what I was wondering about: with both versions in there things may go awry.

I'll try to poke at this myself sometime soon: want to make sure everything works in the new world and in the old one.

qwandor, 2024-04-25:
How are you building the example, and on what git revision? On the latest main branch, if I run `cargo build --features=v2 --target=thumbv7em-none-eabihf` in `examples/v2-speaker` it builds fine.

a-camarillo, 2024-04-25:
I came from the Rust Embedded Discovery Book which uses version `0.12.0` throughout their book and proceeded to work on examples with that version. After removing the `Cargo.lock` file and bumping the `microbit-v2` version to `0.14.0` the build runs fine.

Looking at the history for `0.12.0`, `examples/v2-speaker` was still using `use microbit::hal::prelude::OutputPin;`. I'm guessing that as of the [Update to latest HAL crates](https://github.com/nrf-rs/microbit/commit/cbb158fabd8332048f13b492214124015bf88934), the issue I was running into was fixed.

My apologies, that's my bad as I should have looked more at which version I was using to build.

closed 2024-04-25 by a-camarillo