nrf-rs/microbit #152: MSRV: Bump Rust version in CI?

BartMassey, 2024-07-31:
The latest version of `nrf-hal` has bumped its dependency on the `fixed` crate to a version that requires MSRV Rust 1.78. Shall I go ahead and update the CI infrastructure to use this version of Rust? Should we have a MSRV advertised for this crate?

BartMassey, 2024-08-01:
I guess I'm confused — not uncommon, for sure. If you look at `nrf-hal` you can see that commit a7fed8e3 explicitly bumps `fixed` to version 1.25.1 in `nrf-hal-common/Cargo.toml`, before the change to 0.18. Won't this cause all the users of `nrf-hal` 0.18 to require that version as well?

In any case, let's bump the MSRV in CI here, and also add an MSRV for this crate and check in `Cargo.lock`. If you wouldn't mind doing these I will then revise my other patch.

qwandor, 2024-08-01:
As far as I can see the MSRV change happened in `fixed` 1.28.0 (see their [changelog](https://gitlab.com/tspiteri/fixed#version-1280-news-2024-07-25)). `fixed` 1.25.1 should build fine with an older rustc, and indeed it was building fine here until 1.28.0 was released.

PR #154 (2024-08-01): Run cargo update and bump MSRV.
Fixes #152.
closed 2024-08-01 by qwandor