nrf-rs/microbit #156: cargo install probe-rs-tools flip-link fail on Sonoma 14.6.1

adison, 2024-08-29:
I got below error when running ``` cargo install probe-rs-tools flip-link```
System is Sonoma 14.6.1 (Intel)
Is there anything I can do?
```bash
...
   Compiling fallible-iterator v0.3.0
   Compiling constant_time_eq v0.3.1
   Compiling addr2line v0.22.0
   Compiling svd-parser v0.14.6
   Compiling libtest-mimic v0.7.3
error[E0252]: the name `FromBytes` is defined multiple times
  --> /Users/adison/.cargo/registry/src/index.crates.io-6f17d22bba15001f/probe-rs-0.24.0/src/rtt.rs:59:23
   |
58 | use zerocopy::{AsBytes, FromBytes};
   |                         --------- previous import of the macro `FromBytes` here
59 | use zerocopy_derive::{FromBytes, FromZeroes};
   |                       ^^^^^^^^^--
   |                       |
   |                       `FromBytes` reimported here
   |                       help: remove unnecessary import
   |
   = note: `FromBytes` must be defined only once in the macro namespace of this module

   Compiling insta v1.39.0
   Compiling textwrap v0.16.1
   Compiling ratatui v0.26.3
   Compiling goblin v0.8.2
Compiling indicatif v0.17.8
   Compiling byte-unit v5.1.4
   Compiling defmt-decoder v0.3.11
   Compiling rustyline v14.0.0
   Compiling tracing-appender v0.2.3
   Compiling cargo_metadata v0.18.1
   Compiling itm v0.9.0-rc.1
   Compiling rand v0.8.5
   Compiling directories v5.0.1
   Compiling probe-rs-mi v0.1.0
   Compiling ansi-parser v0.9.1
   Compiling clap_complete v4.5.24
   Compiling cargo-config2 v0.1.27
   Compiling sanitize-filename v0.5.0
   Compiling urlencoding v2.1.3
   Compiling termtree v0.4.1
   Compiling bytesize v1.3.0
   Compiling capstone v0.12.0
For more information about this error, try `rustc --explain E0252`.
error: could not compile `probe-rs` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
  Installing flip-link v0.1.9
    Updating crates.io index
     Locking 10 packages to latest compatible versions
      Adding object v0.35.0 (latest: v0.36.3)
      Adding wasi v0.11.0+wasi-snapshot-preview1 (latest: v0.13.2+wasi-0.2.1)
  Downloaded env_filter v0.1.2
  Downloaded env_logger v0.11.5
  Downloaded 2 crates (44.0 KB) in 0.71s
   Compiling libc v0.2.158
   Compiling log v0.4.22
   Compiling memchr v2.7.4
   Compiling cfg-if v1.0.0
   Compiling env_filter v0.1.2
   Compiling env_logger v0.11.5
   Compiling object v0.35.0
   Compiling getrandom v0.2.15
   Compiling flip-link v0.1.9
    Finished `release` profile [optimized] target(s) in 3m 15s
  Installing /Users/adison/.cargo/bin/flip-link
   Installed package `flip-link v0.1.9` (executable `flip-link`)
error: failed to compile `probe-rs-tools v0.24.0`, intermediate artifacts can be found at `/var/folders/0w/ry2wnfy13bndtc6pnds69g6c0000gn/T/cargo-installUlOT3Q`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
     Summary Successfully installed flip-link! Failed to install probe-rs-tools (see error(s) above).
warning: be sure to add `/Users/adison/.cargo/bin` to your PATH to be able to run the installed binaries
error: some crates failed to install
```

closed 2024-08-29 by qwandor

adison, 2024-08-29:
Got it, thx.

BartMassey, 2024-08-29:
As long as you're here though:

This is part of `probe-rs`'s recent decision to not support installing from `crates.io` anymore. I don't know why they're doing this. For now, you can get things to work with `cargo install --locked`, but I'm told that the latest release to crates.io is considered a mistake and that you should instead install the release binaries via https://probe.rs

BTW, in most situations I think you won't need `flip-link` these days: I have found that just using `rust-lld` as your linker seems to work fine for me.