nrf-rs/microbit #170: docs.rs failed to build microbit-common-0.15.1

Arom1a, 2025-02-09:
I failed to view the source of each items of the crate. 

Here is the build logs to the errors: <https://docs.rs/crate/microbit-common/0.15.1/builds/1312752>.

BartMassey, 2025-02-11:
Looks like `docs.rs` is building for `x86_64`. This won't work, probably. We'll try to figure it out.

mattheww, 2025-02-11:
As I understand it, the build failure happens because `microbit-common` requires either the `v1` or the `v2` feature to be set.

(I get similar output if I just run `cargo doc` for `microbit-common`, but `cargo doc --features=v2` works.)

Looking at https://docs.rs/about/metadata I think the fix would be to put something like the following in `microbit-common`'s `Cargo.toml`.

````
[package.metadata.docs.rs]
features = ["v2"]
````

I'm not sure whether setting the target is necessary too, but it probably wouldn't hurt.

So I've put that in #171 .

BartMassey, 2025-02-11:
As far as I know, `docs.rs` builds with all features set. This may cause its own problems. But it also looked like part of the problems were with the target…

mattheww, 2025-02-11:
I think https://docs.rs/about/metadata says the default is no features:

````
# Features to pass to Cargo (default: [])
features = ["feature1", "feature2"]
````

And the linked build logs show it invoking `cargo +nightly rustdoc` with `--target x86_64-unknown-linux-gnu` but no `--features`.

The log output like
````
failed to resolve: use of undeclared crate or module `hal`
````
matches what I get from `cargo rustdoc` with no features (while if I build with both `v1` and `v2` features it complains that things like `EASY_DMA_SIZE` are defined twice).

closed 2025-02-18 by qwandor