nrf-rs/microbit #172: A safe wrapper around the RADIO peripheral

stanekondrej, 2025-02-14:
Hi, I'm writing a little project of mine that makes use of the RADIO peripheral. Unless I'm blind, it looks like there isn't a safe wrapper for the peripheral at the moment. I would love to implement it, however I just want to ask before I start working on it. I can read datasheets and I think I might be able to do it. Thanks in advance for your response!

stanekondrej, 2025-02-14:
I should probably add that I only own 1.x micro:bits - therefore, I won't be able to test the functionality required to operate the radio peripheral on the 2.x revisions.

stanekondrej, 2025-02-26:
Hi, just an update on this; something happened last week and I've had about 2 hours total to work on this since I opened this issue two weeks ago. I should hopefully be able to make progress with the issue in the evening (it's about 1 AM in my country).

stanekondrej, 2025-02-26:
@mattheww thank you! That would be very kind of you.

stanekondrej, 2025-04-11:
I've finally managed to get around to implementing this. So far, I'm making good progress, however, I'm unsure whether user-facing functions should take `CriticalSections` as parameters, or if they should create them themselves. Does anyone have an opinion on this? I know `Mutex` from `cortex_m` [accepts it as an argument](https://docs.rs/cortex-m/latest/cortex_m/interrupt/struct.Mutex.html#method.borrow), but I just wanted to ask.

mattheww, 2025-04-12:
I don't think there's any precedent here. The existing drivers don't do either of those things.

Most of them aren't high-level enough to care. Their functions just take some wrapped form of the peripheral as parameters, and leave it to the caller to arrange to have unique access to that wrapper.

For example the non-blocking display code provides a function that's intended to be called from a timer interrupt, but doesn't directly help to arrange that. Then there are separate examples showing how to do it using `cortex_m::interrupt::Mutex` and using RTIC.

BartMassey, 2025-04-12:
Yes, when possible it's best for a library crate not to depend on specific synchronization mechanisms (not built into the language): the non-blocking display code is a great example of doing this. There's a lot of synchronization mechanisms in use in Rust Embedded, and any baked-in choice is going to annoy someone who is using something fairly incompatible.

That said, if you need to do sync across APIs, the `critical-section` crate is part of the official HAL, and `portable-atomic` is kind of amazing.

stanekondrej, 2025-04-12:
Okay, thanks for your insight. I'll build the implementation in a synchronization-agnostic way and just add guidance into the doc comments, then. 

One thing I realized is I might have opened an issue in the wrong repo; the 2.4 GhZ radio is common to all the nRF51x (and maybe even nRF52x and whatnot) SoCs, right? Whatever the case, I'm going to leave it here, because the stage I'm in right now doesn't involve actually modifying the board support crate, as I'm writing it as my own "library" that I'll just merge into the upstream crate gradually.

BartMassey, 2025-04-12:
I'm really excited to have your radio library. Thanks for working on this.

I *think* maybe it would be best to just keep this a separate embedded hal library permanently if possible? Then folks could use it with various nRFs, as you suggest. We can add linkage to that library to this BSP crate when the time comes, likely with a feature flag.

Best with the project!